Slicer/SlicerSOFA #14: Consolidate instructions for building Boost as a prerequisites

jcfr, 2024-06-11:
Since both this extension (`Slicer-SOFA`) and `aevaslicer` depend on boost, it may be sensible to document in a more generic location how to locally install boost.

This could possibly be done as an "Infrastructure" wiki page at https://github.com/Slicer/Slicer/wiki

For reference, the process to install Boost is documented at https://gitlab.kitware.com/aeva/aevaslicer/-/issues/10

RafaelPalomar, 2024-06-18:
Thanks @jcfr. I have followed the instructions you refer instead of cloning/compiling boost from repo. There isn't a big differences but it feels quicker as submodules are not pulled one by one. The changes related to this were in RafaelPalomar/Slicer-SOFA#17 and RafaelPalomar/Slicer-SOFA#19

RafaelPalomar, 2024-12-07:
This is addressed in Slicer/SlicerSOFA#28. TLDR: SOFA relies in header-only components of boost; therefore, SlicerSOFA does not need to build and install boost.

closed 2024-12-07 by RafaelPalomar