Slicer/SlicerSOFA #25: Testing and sample data still relying on RafaelPalomar/SlicerSOFATestingData repository

RafaelPalomar, 2024-08-03:
Currently, the sample and testing data used in the extension relies on the `RafaelPalomar/SlicerSOFATestingData` repository.

@jcfr, should we consider moving `RafaelPalomar/SlicerSOFATestingData` to the Slicer organization? perhaps we should upload the data to `Slicer/SlicerTestingData`?

lassoan, 2024-08-03:
Most often we store large binary files for extensions in Girhub Release assets of the extension. Would that work for this case, too?

RafaelPalomar, 2024-08-04:
Sure, that could work too. I suppose I could use the SlicerTestingData software to make the release artifacts for the SlicerSOFA repository?

lassoan, 2024-08-05:
You can use it, it has some nice features, but overall I don't like it that much because the downloaded files have to be manually renamed (if not downloading via Slicer), which 8s not user-friendly.