Slicer/SlicerSOFA #30: Adding new plugins to base Slicer-SOFA

RafaelPalomar, 2024-12-12:
Discussing with @bakpaul, we came up with a list of plugins that would be good to include in Slicer-SOFA by default:

1) [BeamAdapter](https://github.com/sofa-framework/BeamAdapter) - Models of 1D objects, highly used for catheter simulation
2) [Registration](https://github.com/sofa-framework/REGISTRATION) - For model registration purposes.
3) [CGALPlugin](https://github.com/sofa-framework/CGALPlugin/) - Plugin allowing to use cgal for 3D tetrahedral mesh generation
4) [Shell](https://github.com/SofaDefrost/Shell) - implementing 2D elastic model with bending rigidity. Used to rigidify the surface of a model (e.g. liver capsule) of simulate metalic plates.

2) (Q License) and 3) (GPL), due to license issues should  not be included, but 1) and 4) are LGPL and seem to be very useful for medical applications.

lassoan, 2024-12-12:
Running the simulation in a separate process would prevent the viral license to spread to the Slicer application, which would allow using SOFA in commercial applications with GPL plugins. It is good to have the option of running the simulation in the same process, but it would be useful to be able to run it in a separate process, too.

lassoan, 2024-12-12:
In the long term,  https://github.com/BrunoLevy/geogram could be used instead of CGAL.

Registration plugin developer could be asked to change the license to permissive. By now he probably realized that the restrictive license does not do him any good (nobody goes to him to buy a commercial license) while it has clear disadvantages. There may be also many registration librariesthat could be used to reimplement a similar plugin with permissive license.

pieper, 2024-12-12:
If we can just distribute an wrapper, like we do for ffmpeg, and require the user to download and install the GPL parts then we are safe.  But to be on the safe side we can't build/distribute GPL code even if it will be running in a separate process.

bakpaul, 2024-12-13:
I agree with @lassoan on the fact that multiprocess execution must still be in sight of every development, simulation by its nature is crash prone and take a lot of computing time. To enhance hte user experience, we need to target such soft. 

We are the maintainers of Registration, I'll take a look on what is making it GPL and how to solve this ! Maybe we cans add it too at one point !

bakpaul, 2024-12-16:
Here it is : https://github.com/sofa-framework/Registration/pull/18 Registration is now LGPL

RafaelPalomar, 2026-11-10:
Status update: BeamAdapter and Registration (now LGPL) landed and are built by default (`SuperBuild/External_Sofa.cmake`, on `preview`). The Shell and SoftRobots work (closed PR #35 and an unmerged branch) has been dropped as obsolete. CGAL stays out for licensing reasons. Closing. The broader question of letting users bring their own (including GPL) plugins without bundling them is tracked in #32.

closed 2026-11-10 by RafaelPalomar