Slicer/SlicerSOFA #31: Slicer-SOFA extension does not build: `No rule to make target \`packageupload'`

RafaelPalomar, 2024-12-13:
### 🛑 Slicer-SOFA Extension: Package Upload Failure 🛠️

The Slicer-SOFA extension fails to upload the package on both Linux and Windows platforms:

#### 🐧 Linux
🔗 [View Build Error](https://slicer.cdash.org/viewBuildError.php?buildid=3623543)

```
gmake[3]: *** No rule to make target `packageupload'.  Stop.
```

#### 🪟 Windows
🔗 [View Build Error](https://slicer.cdash.org/viewBuildError.php?buildid=3623844)

```
MSBuild version 17.9.8+b34f75857 for .NET Framework

MSBUILD : error MSB1009: Project file does not exist.

Switch: packageupload.vcxproj
```

---

❓ **Could the fact that the extension and one of its modules share the same name be causing a target conflict?**

lassoan, 2024-12-13:
> Could the fact that the extension and one of its modules share the same name be causing a target conflict?

It is not a problem. For example, TotalSegmentator is the name of the extension and module.

I've tried the build locally now and I was able to reproduce the issue. `Project file does not exist` issue was reported because the inner build was not even configured because SOFA build failed with this error (probably we don't see this error in the CDash log because build log of external libraries are not uploaded):

```
12>C:\Program Files\Microsoft Visual Studio\2022\Community\MSBuild\Microsoft\VC\v170\Microsoft.CppBuild.targets(382,5): error MSB3491: Could not write lines to file "x64\Release\SofaGeneralExplicitOdeSolver_relocatable_install\SofaGene.21A0408A.tlog\SofaGeneralExplicitOdeSolver_relocatable_install.lastbuildstate". Path: x64\Release\SofaGeneralExplicitOdeSolver_relocatable_install\SofaGene.21A0408A.tlog\SofaGeneralExplicitOdeSolver_relocatable_install.lastbuildstate exceeds the OS max path limit. The fully qualified file name must be less than 260 characters.  [C:\D\EI-build\SlicerSOFA-build\Sofa-build\applications\collections\deprecated\modules\SofaGeneralExplicitOdeSolver\SofaGeneralExplicitOdeSolver_relocatable_install.vcxproj]
12>C:\Program Files\Microsoft Visual Studio\2022\Community\MSBuild\Microsoft\VC\v170\Microsoft.CppBuild.targets(382,5): error MSB3491: Could not write lines to file "x64\Release\SofaGeneralImplicitOdeSolver_relocatable_install\SofaGene.DB75AD95.tlog\SofaGeneralImplicitOdeSolver_relocatable_install.lastbuildstate". Path: x64\Release\SofaGeneralImplicitOdeSolver_relocatable_install\SofaGene.DB75AD95.tlog\SofaGeneralImplicitOdeSolver_relocatable_install.lastbuildstate exceeds the OS max path limit. The fully qualified file name must be less than 260 characters.  [C:\D\EI-build\SlicerSOFA-build\Sofa-build\applications\collections\deprecated\modules\SofaGeneralImplicitOdeSolver\SofaGeneralImplicitOdeSolver_relocatable_install.vcxproj]
```

So, on Windows, the issue is due to too long paths. Maybe shortening `relocatable_install` to something like `relinst` could be just enough.

On Linux/macOS, the error seems to be different. I cannot investigate those.

lassoan, 2024-12-13:
I've submitted a PR for the long path issue on Windows (https://github.com/Slicer/sofa/pull/1) and the incorrect extension description json file that causes the packageupload problem on all platforms (https://github.com/Slicer/ExtensionsIndex/pull/2132).

lassoan, 2024-12-13:
@RafaelPalomar please review both PRs then I'll merge them. After that please update the git hash of sofa in the SlicerSOFA superbuild and I'll review&merge.

RafaelPalomar, 2024-12-16:
Closing after Slicer/ExtensionsIndex#2132

> After that please update the git hash of sofa in the SlicerSOFA superbuild and I'll review&merge.

For reference there is Slicer/SlicerSOFA#34

closed 2024-12-16 by RafaelPalomar