Slicer/SlicerSOFA #44: Volumetric model generated by SegmentMesher cannot deform

chz31, 2025-02-12:
I used SegmentMesher to make a vol model in vtk unstructured grid format of the eye soft tissue. However, I could not use softTissueSimulation to deform it (attached video). The lung model from the sample data can be simulated well. 

No error was printed out.

The vol model is here:
[eye soft tissue model](https://drive.google.com/file/d/1Grwh5WnGhbf0Uycii9pV3fb0orWGY6gP/view?usp=drive_link)

The SlicerSOFA is built with Slicer 5.7.0 in Ubuntu 22.04.

https://github.com/user-attachments/assets/632f6c91-8b63-4be6-911d-b44b228f691d

RafaelPalomar, 2025-02-13:
Thank you @chz31 for the report. We will have a look at it

RafaelPalomar, 2025-02-19:
@bakpaul, I'm having a look a this issue. Could the fixed parameters of the simulation produce this effect? The volume, mass, size,.. of this geometry is very different from that of a lung. Could `youngModulus` and `poissonRatio` parameters of the TetrahedronFEMForceField be playing a role here?

bakpaul, 2025-02-19:
I won't change those variable during the simulation. Some stuff are precomputed and would require a call to init to be taken into account and I would not guarantee that this is supported. It is better ot make sure the value are right before launching the simulation. 

If the scale is different (for instance you are in m and the youg modulus has been set for a model in mm) then yes, you might have 13^3 time too much young modulus.

chz31, 2025-02-19:
@RafaelPalomar @bakpaul The original volumetric model was in msh format created by gmsh and could run simulation in SOFA. However, I could not load the msh model into Slicer, so I made a vtk unstructured grid model using the SegmentMesher module in Slicer (sorry I forgot the settings!). Here are the original msh file & script I used for SOFA: [link](https://drive.google.com/drive/folders/19OjwGkUXrimA9s9lgiTawZaYMc88vQlz?usp=sharing). I used the fine visual obj model to create a vtk unstructured grid model.

The settings for the orbit soft tissue mechanical model that can successfully run the simulation is: 'TetrahedronFEMForceField', name='FEM', youngModulus=5e4, poissonRatio=0.47, method='large'

Is there anyway to load vtk unstructured grid in SOFA as well so I can try the settings?

RafaelPalomar, 2025-02-20:
@chz31, I don't think it is a matter of the way the model was computed; this looks like a valid model to me. In fact, the gravity has its effect on the simulation if enabled.

RafaelPalomar, 2025-02-20:
> I won't change those variable during the simulation. Some stuff are precomputed and would require a call to init to be taken into account and I would not guarantee that this is supported. It is better ot make sure the value are right before launching the simulation.

Right. I can expose some of these values to the ui and make sure that they can only be modified when the simulation has stopped and need a restart. @bakpaul, which values do you think it makes sense to expose? This should give users the possibility to use new models.

> If the scale is different (for instance you are in m and the youg modulus has been set for a model in mm) then yes, you might have 13^3 time too much young modulus.

@bakpaul, would it be possible for you to give a try to this simulation in SOFA with the picker? It looks strange that in this model there is no effect.

chz31, 2025-05-12:
> Right. I can expose some of these values to the ui and make sure that they can only be modified when the simulation has stopped and need a restart. [@bakpaul](https://github.com/bakpaul), which [...]

Hi Rafael,

Sorry for following up this issue late. I just installed the SlicerSOFA extension directly from the Extension Manager and tested the same model. It appears that I still could not deform the model.

Here is my original [SOFA script](https://drive.google.com/file/d/1ZQrjU6UMzyvgZs5hhoZJYxFWg3NqdTFx/view?usp=sharing) that can deform the model. I checked the `SoftTisssueSimulation.py` and found the methods are basically the same. 

The only difference is that I used  `eyeNode.addObject('TetrahedronSetGeometryAlgorithms', template='Vec3d', name='GeomAlgo')` instead of `(TetrahedronSetTopologyContainer, name = 'Container', position=None, tetrahedron=None)`. I also set `youngsmodulus = 5e5`, and `poisson ratio = 0.47`, but in SOFA, the values did not matter for the simulation to occur.

RafaelPalomar, 2025-05-16:
@chz31, I suspect we are not transferring correctly these parameters. I'll try to have a look at it next week, but we are still updating some other parts of SlicerSOFA that have priority now.

chz31, 2025-05-17:
@RafaelPalomar Thank you very much. I also tried to use SegmentMesher (Default Setting, Cleaver) to get a liver volumetric model from the LiverSimulationScene in the sample data. It also did not work.

[image]

It also sometimes reported below error:
```
Traceback (most recent call last):
  File "/home/chizhang/Slicer-5.9.0-2025-02-07-linux-amd64/slicer.org/Extensions-33462/SlicerSOFA/lib/Slicer-5.9/qt-scripted-modules/SoftTissueSimulation.py", line 336, in startSimulation
    self.logic.startSimulation()
  File "/home/chizhang/Slicer-5.9.0-2025-02-07-linux-amd64/slicer.org/Extensions-33462/SlicerSOFA/lib/Slicer-5.9/qt-scripted-modules/SoftTissueSimulation.py", line 402, in startSimulation
    super().startSimulation()
  File "/home/chizhang/Slicer-5.9.0-2025-02-07-linux-amd64/slicer.org/Extensions-33462/SlicerSOFA/lib/Slicer-5.9/qt-scripted-modules/SlicerSofa.py", line 356, in startSimulation
    self.onSimulationStarted()
  File "/home/chizhang/Slicer-5.9.0-2025-02-07-linux-amd64/slicer.org/Extensions-33462/SlicerSOFA/lib/Slicer-5.9/qt-scripted-modules/SlicerSofa.py", line 376, in onSimulationStarted
    self.ui.updateWidgetOnSimulation()
  File "/home/chizhang/Slicer-5.9.0-2025-02-07-linux-amd64/slicer.org/Extensions-33462/SlicerSOFA/lib/Slicer-5.9/qt-scripted-modules/SlicerSofa.py", line 235, in updateWidgetOnSimulation
    (self._parameterNode.isSimulationRunning and not disable) or
AttributeError: 'NoneType' object has no attribute 'isSimulationRunning'
```

chz31, 2025-06-21:
Hi Rafael @RafaelPalomar,

Just posting some tests here ahead of the project week. I tested deforming the same orbital soft tissue model using the gravity and moving point functions from `SoftTissueSimulationTest`. I only changed ROI and moving points by customized points.

```
fixedROINode = slicer.util.getNode('fixedROI')

# Set initial position of the moving point using logic
startPtNode = slicer.util.getNode('startPt')

parameterNode = logic.getParameterNode()
parameterNode.modelNode = simulationModelNode
# logic.addMovingPoint()
movingPointNode = startPtNode
# movingPointNode = logic.getParameterNode().movingPointNode

# Define start and end positions for the moving point
startPosition = [0, 0, 0]
endPosition = [0, 0, 0]

endPtNode = slicer.util.getNode('endPt')

startPtNode.GetNthControlPointPosition(0, startPosition)
# startPosition = list(movingPointNode.GetNthControlPointPosition(0))
endPtNode.GetNthControlPointPosition(0, endPosition)

# Calculate step size for linear interpolation
totalSteps = 100
interpolationStep = [(end - start) / totalSteps for start, end in zip(startPosition, endPosition)]

# self.delayDisplay("Setting up simulation parameters")
# Assign simulation parameters to the parameter node
parameterNode.boundaryROI = fixedROINode
parameterNode.movingPointNode = movingPointNode
parameterNode.gravityMagnitude = 0  # Disabling gravity
parameterNode.dt = 0.01
parameterNode.currentStep = 0
parameterNode.totalSteps = totalSteps

# self.delayDisplay("Starting moving point-only simulation")
# Start the simulation and render view
logic.startSimulation()
view = slicer.app.layoutManager().threeDWidget(0).threeDView()

# Run simulation steps with gradual movement of the point
for step in range(parameterNode.totalSteps):
    # Update the point's position by adding the interpolation step to the current position
    new_position = [
        startPosition[0] + step * interpolationStep[0],
        startPosition[1] + step * interpolationStep[1],
        startPosition[2] + step * interpolationStep[2]
    ]
    movingPointNode.SetNthControlPointPosition(0, *new_position)

    # Advance simulation and render
    logic.simulationStep()
    view.forceRender()

# Stop the simulation and clean up
logic.stopSimulation()
```

Again, gravity worked well and the model can deform. However, the moving points still could not deform the model. The start point itself did move to the end point, though.

[image]

RafaelPalomar, 2025-06-21:
@chz31, I have added this point explicitly on the agenda for ProjectWeek. We will have the right people in the room to solve this issue.

RafaelPalomar, 2025-06-28:
@chz31, can we close this on the base of the demo you got working on PW?

closed 2025-06-30 by chz31