Slicer/SlicerSOFA #45: Extension not installing on mac for Slicer 5.8.0

pieper, 2025-03-02:
[image]

```
Retrieving extension metadata for SlicerSOFA extension
Retrieving SlicerSOFA extension files (extensionId: 679375cf1357655fd58628ed)
Downloading SlicerSOFA extension (item_id: 679375cf1357655fd58628ed, file_id: 67a8dc6c133f76d338e792e4)
ctk::copyDirRecursively: Failed to copy files from  "/Applications/Slicer-5.8.0.app/Contents/Extensions-33216/SlicerSOFA-XXXXXX/lib/Slicer-5.8/qt-scripted-modules/Resources/Icons"  into  "/Applications/Slicer-5.8.0.app/Contents/Extensions-33216/SlicerSOFA/lib/Slicer-5.8/qt-scripted-modules/Resources/Icons"
ctk::copyDirRecursively: Failed to copy files from  "/Applications/Slicer-5.8.0.app/Contents/Extensions-33216/SlicerSOFA-XXXXXX/lib/Slicer-5.8/qt-scripted-modules/Resources"  into  "/Applications/Slicer-5.8.0.app/Contents/Extensions-33216/SlicerSOFA/lib/Slicer-5.8/qt-scripted-modules/Resources"
ctk::copyDirRecursively: Failed to copy files from  "/Applications/Slicer-5.8.0.app/Contents/Extensions-33216/SlicerSOFA-XXXXXX/lib/Slicer-5.8/qt-scripted-modules"  into  "/Applications/Slicer-5.8.0.app/Contents/Extensions-33216/SlicerSOFA/lib/Slicer-5.8/qt-scripted-modules"
ctk::copyDirRecursively: Failed to copy files from  "/Applications/Slicer-5.8.0.app/Contents/Extensions-33216/SlicerSOFA-XXXXXX/lib/Slicer-5.8"  into  "/Applications/Slicer-5.8.0.app/Contents/Extensions-33216/SlicerSOFA/lib/Slicer-5.8"
ctk::copyDirRecursively: Failed to copy files from  "/Applications/Slicer-5.8.0.app/Contents/Extensions-33216/SlicerSOFA-XXXXXX/lib"  into  "/Applications/Slicer-5.8.0.app/Contents/Extensions-33216/SlicerSOFA/lib"
Failed to copy directory /Applications/Slicer-5.8.0.app/Contents/Extensions-33216/SlicerSOFA-XXXXXX into directory /Applications/Slicer-5.8.0.app/Contents/Extensions-33216/SlicerSOFA
```

pieper, 2025-03-02:
Checked and the same issue happens for Slicer-5.9.0-2025-03-01-macosx-amd64

RafaelPalomar, 2025-03-03:
@pieper, thanks for reporting I'll check on this.

RafaelPalomar, 2025-03-31:
This error originates  in the installation (packaging) of the extension. The current structure for MacOS looks like this:

```
Slicer.app
lib
plugins
share
collections
README.md
....
```

it seems that in Mac we can only preserve `Slicer.app` and [FHS-compliant directories](https://en.wikipedia.org/wiki/Filesystem_Hierarchy_Standard) (I haven't found any related information about this fact, but it seems to be the case, empirically).

Slicer/SlicerSOFA#46 is on its way to solve this issue. In order to finish it, we need to modify 3 external plugins:
  - SlicerPython3, which is included in the Slicer organization
  - SofaIGTLink (not included in Slicer organization)
  - STLIB (not included in Slicer organization)

@pieper, @jcfr, there are two ways around this. (1) We could fork `SofaIGTLink` and `STLIB` in the Slicer organization. (2) Another approach could be a simple one-line string replacement done through `External_SOFA.cmake` until this approach is accepted upstream. What do you think it would be best? 

Slicer/SlicerSOFA#46 would need to be tested on Win/Mac/Linux to make sure these changes are not breaking anything.

closed 2025-07-29 by RafaelPalomar