Slicer/SlicerSOFA #56: FixedROI box uninitialized when `position` refers to `rest_position` of a mechanical object built by `SparseGridTopology`

RafaelPalomar, 2025-09-30:
The following Warning is triggered when running the simulation scene below.

```
[WARNING] [BoxROI(FixedROI)] No rest position yet defined. ROI might not work properly. 
This may be caused by an early initialization of the ROI before  
the mesh or the MechanicalObject of the node was initialized too
```

The `BoxROI` does not seem to have any effect. It looks to me that relying on the geometry/topology created by `SparseGridTopology` makes the `BoxROI` not ready at initialization. The same scene in SOFA v24.06 used to work.

https://github.com/Slicer/SlicerSOFA/blob/aea391917b6d065370e1942b55e8cf5407ebca93/SparseGridSimulation/SparseGridSimulation.py#L168-L202

@bakpaul, do you have any pointers? Does this happen in other SOFA simulations?

@eleonore2001, for reference. This is probably why our simulation results don't seem to deform the liver. The organ just flies out because there is no anchoring points.

bakpaul, 2025-09-30:
I don't know about any behavior change between the v24.06 and the recent version of SOFA on the boxROI. 
But you might be able to fix this issue by setting by hand the input values of the MO. Try adding `position="@../InputSurfaceNode/Container.position", rest_position="@../InputSurfaceNode/Container.position"` to the MO creation call

bakpaul, 2025-09-30:
Ok, let me try to recreate the error. In the meantime, I remember that you had your own way of initializing the root node in SlicerSOFA. We've already discussed it but I cannot remember exactly what was wrong but it wasn't the way it was supposed to be. Could you take a look at that ? Make sure you are not initializing the root at the wrong moment, before its creation or before every data is accessible.

RafaelPalomar, 2025-09-30:
That was it! We may have introduced that regression along the process of updating. Thank a lot @bakpaul!